trikset/trik-studio #25: Update main README.md

iakov, 2019-02-24:
1. Mention/reference old https://github.com/qreal/qreal project. 
1. Remove other mentions.
1. Badges,
1. TRIK Studio logo (coming soon)
1. Reference to issue reporting procedure REPORT_ISSUE.md.
1. Explain build requirements and build steps (or BUILD.md)

iakov, 2019-04-20:
We need build procedure description. Issue reporting can be postponed as a separate issue

IKhonakhbeeva, 2019-12-10:
Инструкция к винде (предварительная) :
https://github.com/IKhonakhbeeva/trik-studio/wiki/Windows-build-with-Qt-Creator

IKhonakhbeeva, 2019-12-10:
Шаблоны по issue - пока что просто не знаем, как он должен выглядеть

kleo-53, 2023-09-21:
Wrote instructions for building the TRIK studio for 32-bit Windows from console
https://github.com/kleo-53/trik-studio/wiki/Windows-x32-build-with-console

iakov, 2023-09-22:
#1681 merged