trikset/trik-studio #27: Opening the lastSafeCode.js when opening another js project but not saved previous js project.

ilyashir, 2019-02-24:
Когда js проект не сохранен и открываются другие js проекты, тогда с ними же открывается файл lastSaveCode.js с содержанием первого открытого несохраненного проекта.

ilyashir, 2019-02-24:
[image]

IKhonakhbeeva, 2019-05-06:
Не получилось воспроизвести.
Возможно, неправильно поняла.
@ilyashir , недавно произошли изменения, связанные с открытием/закрытием файлов. На последней версии баг сохранился? Если да, можно немного более подробную инструкцию, как можно повторить баг?

IKhonakhbeeva, 2019-05-06:
@ilyashir , please try version from #164. You can get it from appveyorBot's comment.

PR #164 (2019-05-07): Correct open lastSavedCode
Fixed #27
closed 2019-05-07 by iakov