trikset/trik-studio #130: "Read RGB to Variable" from raw to 0..255

anastasiia-kornilova, 2019-04-20:
В блоке Read RGB to Variable привести значения из сырых в диапазон 0..255.

anastasiia-kornilova, 2019-06-06:
Min and max values and modes can be found here -- https://github.com/mindboards/ev3sources-xtended/blob/master/ev3sources/lms2012/lms2012/Linux_AM1808/sys/settings/typedata.rcf

anastasiia-kornilova, 2019-06-06:
In documentation raw rgb values from 0 to 1023, but LED in sensor is not so bright to reach higher border. http://blog.thekitchenscientist.co.uk/2015/02/ev3-python-raw-rgb-values.html -- here is research on values for different colors, they are not bigger than 500.

anastasiia-kornilova, 2019-06-10:
От safilippov: "Предлагаю воспользоваться примерами из RobotC и EV3 Software, посмотреть их диапазоны и сделать по аналогии."

IKhonakhbeeva, 2019-11-19:
RobotC приводит значения к 0..255, но каким именно способом - найти не получилось (исходников в открытом доступе вроде нет)
EV3 Software вообще этого режима по умолчанию не имеет, есть какой-то кастомный блок, но без документации (но есть примечание, что значения "обычно от 0 до 800", то есть вероятнее всего просто сырые)

Как именно хочется приводить к 0..255?
Можно просто домножать на константу ("повышать яркость"), обрезая до 255
Или брать долю от максимального значения среди трех. Например, (5, 0, 100) -> (5/100 * 255, 0, 100/100 * 255) = (13, 0, 255)

iakov, 2019-11-20:
Надо ставить эксперименты с RobotC.

IKhonakhbeeva, 2019-12-09:
На robotC на практически белый максимум выдавал 90 из 255, но в основном колебался в районе 70-80. Но тут не очень светло, так что может быть в яркой комнате на хороший белый цвет он реагировал бы и сильнее, но в любом случая явно не 255. То есть тут тупое приведение из 1024 к 256 делением на 4, судя по всему.

IKhonakhbeeva, 2019-12-09:
Данные напрямую с робота в этом же помещении на том же листе действительно примерно в четыре раза больше и все