trikset/trik-studio #140: Не работает печать.

ilaforer, 2019-04-22:
[image]

iakov, 2019-05-12:
На линуксе нормально работает. Печатает в PDF и на принтер. Перейдём на Qt 5.12, потом посмотрим.

IKhonakhbeeva, 2019-05-24:
Винда с 5.12 работает, но сетка разъезжается
[image]

А вот проблемы с размером шрифта и на линуксе проглядывают
[image]

iakov, 2019-05-24:
А при печати в PDF всё нормально или проблема воспроизводится?

IKhonakhbeeva, 2019-05-24:
На винде даже не вижу пункта "печать в пдф"
Посмотрю, поиграюсь с настройками, вроде как работа с принтером идет примитивно стандартно-кутешная, так что странно

IKhonakhbeeva, 2019-05-28:
На линуксе и макосе QPrintDialog используется системный, который не знает о кутешной возможности печатать в файл. Так что нужна отдельная кнопка, если печать в пдф необходима.

> On Windows and macOS, the native print dialog is used, which means that some QWidget and QDialog properties set on the dialog won't be respected

При печати в пдф (винда) поведение аналогично линуксу.

PR #231 (2019-05-29): Print
Fixed #140 
Fixed #145
closed 2019-05-29 by iakov