trikset/trik-studio #143: Win7 - нет платформы TRIK

ilaforer, 2019-04-22:
Программы открываются, но такой платформы нет в настройках.
При открытии программ для TRIK пропадает выбор модели 2D, генерации и т.д.

[image]

IKhonakhbeeva, 2019-04-26:
Please try the latest version
[image]

IKhonakhbeeva, 2019-04-27:
На чистой машине баг тоже не воспроизводится

ilaforer, 2019-04-29:
Надо уточнить у Ильи на какой версии это было. Он нашел этот баг.

ilaforer, 2019-04-29:
Это было на какой-то бете. В последней версии проблема не появляется.

iakov, 2019-05-24:
@IKhonakhbeeva надо проверить 2019.1 в чистом окружении Win7.

PR #218 (2019-05-24): Get python dlls from Python directory
Fixed #143
closed 2019-05-24 by iakov