trikset/trik-studio #146: Описание блока "ждать датчик касания" на английском

ilaforer, 2019-04-22:
[image]

iakov, 2019-04-25:
@IKhonakhbeeva , проверь ещё и у других конструкторов, мб ещё где забыли обновить перевод.

IKhonakhbeeva, 2019-05-02:
Вывести все файлы с незаконченным переводом можно следующим образом (возможно, можно и красивее, но пока так):
`cd qrtranslations/ru && grep -r -l 'translation type="unfinished"'`

IKhonakhbeeva, 2019-05-06:
Ха. Все куда интереснее. Некоторые файлы перевода ([ev3|nxt|trik]Metamodel_ru.ts) никак вообще не связаны сейчас с проектом, просто по сути болтаются рядом. Ни один TRANSLATIONS не указывает в [ev3|nxt|trik]Metamodel_ru.ts. Соответственно, lupdate не знает, что ему надо вообще это обновлять. А lrelease работает за счет того, что он взаимодействует не с проектом, а с файлами ts, которые остались как артефакты.

Так что факт того, что не поехало все давно - ужасная случайность.

Добавить просто TRANSLATIONS не помогает, т.к. нужно переводить включения в xml-ки, достаточно странным образом (с точки зрения qt) подключенных к проекту.

Оказалось, не настолько пустецкий issue, буду разбираться

IKhonakhbeeva, 2019-05-06:
Ого. Интересный механизм)
В общем, lupdate надо запускать на всякие translations.pro (которые никак к проекту не привязаны, как можно было заметить, а просто болтаются рядышком..)
Связано с тем, что нужно включать сгенерированные файлы, а их нехорошо включать в основном проекте.

Так что надо будет подправить проверку lupdate на трависе + мб написать именно скрипт, который все lupdate бы делал

iakov, 2019-05-06:
> Добавить просто TRANSLATIONS не помогает, т.к. нужно переводить включения в xml-ки, достаточно странным образом (с точки зрения qt) подключенных к проекту.

Что означает эта фраза?
Почему переводы не связаны с проектом? Разве это нормально?

IKhonakhbeeva, 2019-05-06:
Да, это нормально.

Вся фича с translations.pro заключается в том, чтобы после сборки добавить переводы сгенерированных файлов, которых нет в репо. Этот lupdate никак нельзя сделать до сборки, потому что он просто не найдет сгенерированный файлы.

IKhonakhbeeva, 2019-05-06:
Но есть одна очень большая проблема.
Файлы на линуксе и на винде генерируются разные.
(не в плане общего содержимого, а именно строчки гуляют)
Поэтому, видимо, нужно как-то делать lupdate + lrelease (чтобы эти новые переводы перетащить в бины) перед каждой сборкой инсталлера. Ну или вообще в конце каждого билда.

IKhonakhbeeva, 2019-05-06:
> Что означает эта фраза?
Просто быстрая попытка добавить TRANSLATIONS = ../../../../qrtranslations/puugins/robots/ev3Metamodel_ru.ts
Что сказало бы lupdate-у, что "вот файл для переводов, переводи туда"
После вызова lupdate в файле все переводы стали "устаревшими". Так как: 1) сначала подумала, что переводить надо xml, а это надо делать не совсем стандартным образом; 2) потом поняла, что надо переводить сгенерированные сборкой файлы, так что переводить ему было по факту нечего (а даже если бы эти файлы и были, в ev3Metamodel.pro не включены генерирующиеся файлы как headers или source).

iakov, 2019-05-07:
Откладываем пока в сторону, потом обсудим. Из текста выше ничего не понял. Пока вижу только, что при генерации используются для промежуточного представления структуры данных, которые не гарантируют порядка (HashMap ?). Сначала надо добиться стабильности в генераторах. Что делать с переводами тоже надо решать, но не в текущей версии 19.04

IKhonakhbeeva, 2019-05-07:
В качестве быстрого фикса - можно на аппвейере эти переводы обновлять, тогда хотя бы на серваке будет правильный перевод

PR #169 (2019-05-12): qrxc stable order magic for lupdate
Fixed #146
closed 2019-05-12 by IKhonakhbeeva