trikset/trik-studio #147: В TRIK Studio 3.3.0-rc8 не открывается сейв alongTheLine. Студия просто вылетает.

ilaforer, 2019-04-22:


ilaforer, 2019-04-29:
Регресс относительно 3.2.0

IKhonakhbeeva, 2019-05-01:
alongTheLine имеет две диаграммы, если одну из них удалить (открыв в 3.2.0), то получившаяся qrs-ка открывается последней версией

IKhonakhbeeva, 2019-05-01:
Что это за странные "моторы вперед" на второй диаграмме? Подсвечиваются серым, а если их заменить на те, которые есть в списке - попытка открыть крашится позже.

IKhonakhbeeva, 2019-05-02:
Если установлен nxt, то все работает

IKhonakhbeeva, 2019-05-02:
3.2.0 в инсталлере по умолчанию ставит nxt, без них на 3.2.0 так же падает

iakov, 2019-05-06:
- [ ] Для фикса -- только обновить qrs в примерах. Осторожно с локалью, на которой делается qrs: 
 лучше английская.
 - [ ] Отдельно надо разобраться, почему креш, а не сообщение об ошибке, что не установлен плагин. Можно сделать отдельным issue, это можно НЕ в 19.04
 - [ ] Проблемы инсталлятора, что он плохо добавляет плагины -- отдельным issue, и не 19.04

PR #163 (2019-05-07): Remove diagram with nxt motors from trik examples
Fixed #147 
Hot fix.

Основная проблема - #162
closed 2019-05-07 by iakov