trikset/trik-studio #148: Unattended install is impossible

iakov, 2019-04-25:
We are missing a solution for classroom support: an unattended/automated installation of a new version. Seems like it is possible with a js script: https://stackoverflow.com/questions/23564377/how-can-i-install-qt-5-2-1-from-the-command-line-in-cygwin

IKhonakhbeeva, 2019-05-22:
* [ ] unattended install qt (on linux)
* [ ] unattended install trik-studio on windows (appveyor)
* [ ] unattended install trik-studio on windows with uninstall previous version

iakov, 2019-05-22:
Надо подумать о том, можем ли мы хотя бы скрипт `reinstall-trik-studio.cmd` сделать, где сначала старую удаляем, потом новую ставим.

IKhonakhbeeva, 2019-05-23:
Да, с uninstall.qs все достаточно просто

iakov, 2019-05-23:
Ситуация 1: один человек обслуживает класс компов. Нужно, чтобы он что-то запустил и без нажатия на кнопки в окнах инсталлера получил установленную свежую студию. Пока она ставится на одном компе, человек переходит к следующему. Решение примерно такое -- флешка или папка на сетевом диске. В ней всё, что нужно. На флешке даже `autorun.inf`, где вызывается `reinstall-trik-studio.cmd`. 
1. Что должно быть в папке/на флешке ещё?
1. Что должно быть примерно в этом .cmd?

Усложнённая ситуация -- обновления ставятся с сетевого диска автоматически. Решение: при загрузке компа монтируется диск и с сетевого диска вызывается "check-trik-studio.cmd", а этот cmd находит студию и проверяет, что версия НЕ отличается от той, какую он автоматически поставит сейчас. И если отличается, то сводим к предыдущей задаче, которую надо сначала решить.

Про сетевые диски -- это уже потом, тут важно научиться проверять версию установленной Студии. Давай сначала решим основную задачу -- ручное обновление. Вторую подзадачу можно вынести в отдельный issue

PR #237 (2019-05-31): Add reinstall script
Fixed #148
closed 2019-05-31 by iakov