trikset/trik-studio #180: Смещается палитра и редактор свойств при открытии студии

ilaforer, 2019-05-14:
[image]

Воспроизводится на квадратных мониках и на вин 7 и на вин 10. Лечится вот этой кнопкой 

[image]

ilaforer, 2019-05-14:
Должно быть вот так 
[image]

iakov, 2019-05-22:
После #177 посмотрим. Нужно уметь воспроизводить ещё стабильно этот баг.

iakov, 2019-05-24:
@ilaforer , посмотри на новом билде, пожалуйста. Воспроизводится ли https://dl.trikset.com/ts/fresh/installer/trik-studio-installer-master-i686.exe Это сложно отлавливаемая проблема.

IKhonakhbeeva, 2019-05-31:
Воспроизводится стабильно на виртуалке с квадратными настройками экрана

[image]
После "свернуть в окно - развернуть":
[image]

iakov, 2019-05-31:
@IKhonakhbeeva ,  не бывает "квадратных настроек", бывает "выставлен такой-то графический режим". Какой режим?

IKhonakhbeeva, 2019-05-31:
Разрешение 800х600
Ситуация как на первых скринах в этом ишью - 1280х1024

PR #348 (2019-07-02): Fix palette on square screens
Fix #180
closed 2019-07-02 by iakov