trikset/trik-studio #223: Braking mode in motors blocks

anastasiia-kornilova, 2019-05-25:
Ввести в свойства всех блоков управления моторами (вперед, назад, стоп) свойство
«Режим торможения». По умолчанию он должен быть включен.
Противоположность – «плавающий (скользящий) режим».

iakov, 2019-05-31:
Ещё есть  #117 , ну да ладно
И оно же #227 , похоже

iakov, 2019-05-31:
Надо не забыть интепретацию в 2D в режиме реальной физики и интерпретацию JS для ТРИК.  Но можно это разнести по разным задачам потом, а сначала просто сделать блок.

iakov, 2019-06-16:
Блок в ТРИКе  теперь есть. Надо делать  для ev3 -- #227

iakov, 2019-07-02:
Надо бы проверить для NXT,  тогда останется сделать только для режима имитационного моделирования.

iakov, 2020-10-09:
Что у нас тут осталось доделать? В каких-то конструкторах в имитационном или в интерпретации/генерации что-то не сделано ещё?