trikset/trik-studio #225: 2D model move moving object

anastasiia-kornilova, 2019-05-25:
Исправить ошибку, при которой робот упирается в подвижные объекты и не может
их сдвинуть в 2D.

iakov, 2019-05-25:
Same? https://github.com/trikset/trik-studio/issues/187

iakov, 2019-06-17:
Это без реалистичной физики. Нужно посмотреть в 2D модели для EV3, NXT, TRIK. Робот должен просто толкать банки.

IKhonakhbeeva, 2019-07-04:
Полученная информация, чтобы не разбираться в ишью с нуля заново:

В plugins\robots\common\twoDModel\src\engine\model\model.cpp есть mTimeline - отвечает за время, по сигналу tick() происходит пересчет параметров, по nextFrame() - перерисовка.

Для пересчета по tick() в SimplePhysicsEngine::recalculateParameters необходимо добавить пересчет параметров не только для роботов (опираться можно на SimplePhysicsEngine::recalculateParameters(qreal timeInterval, RobotModel &robot)). Для этого необходимо будет добавить в SimplePhysicsEngine средства хранения промежуточных (не перерисованных до nextFrame) посчитанных изменений. То есть аналог mPositionShift и mRotation для роботов в SimplePhysicsEngine, или аналог mBox2DDynamicItems для всех итемов из Box2DPhysicsEngine (реалистичной физики).

Для перерисовки по nextFrame() необходимо:
1) Реализовать nextFrame в SimplePhysicsEngine (можно опираться на Box2DPhysicsEngine::nextFrame).
2) законнектить nextFrame в Model::initPhysics
`connect(&mTimeline, &Timeline::nextFrame, this, [this](){ mSimplePhysicsEngine->nextFrame();	})`

Для адекватного поведения робота (не отталкивание от мяча, как от стенки) так же надо подправить обработку с checkCollision в SimplePhysicsEngine::recalculateParameters(qreal timeInterval, RobotModel &robot)

На первый взгляд - основная проблема в том, что checkCollision не подходит для выяснения, в какую сторону двигаться при столкновении (робот отскакивает просто в обратную сторону от направления своего движения, т.е. важно наличие скорости у объекта, что явно не подходит для полноценной обработки столкновений).

Полезно использовать принадлежность BallItem, RobotItem, SkittleItem к SolidItem, благодаря чему можно получить массу объекта и прочие полезные штуки.

iakov, 2019-07-07:
Имеет смысл сделать режим "реалистичной физики" удобными и понятным для пользователя, а не заниматься усовершенствованием простого движка. Postponed пока не появятся контраргументы.