trikset/trik-studio #226: Gyro implementation in 2D

anastasiia-kornilova, 2019-05-25:
Реализовать поддержку гиро-датчика в 2D.

IKhonakhbeeva, 2019-07-05:
Проблема следующая:
Ev3-шный гироскоп [наследуется](https://github.com/trikset/trik-studio/blob/master/plugins/robots/common/ev3Kit/include/ev3Kit/robotModel/parts/ev3Gyroscope.h#L23) от ScalarSensor
Базовый же [наследуется](https://github.com/trikset/trik-studio/blob/master/plugins/robots/common/kitBase/include/kitBase/robotModel/robotParts/gyroscopeSensor.h#L25) от VectorSensor

Чтобы воспринимать его как девайс-гироскоп в 2d, надо чтобы ev3-шный [наследовался](https://github.com/trikset/trik-studio/blob/master/plugins/robots/common/kitBase/src/robotModel/deviceInfo.cpp#L49) от базового (возможно через несколько классов).
Так что сейчас в 2d модели даже не считываются значения с гироскопа. И компаса, по абсолютно аналогичным проблемам. Что в логах отображается как `Can not create device "ev3::robotModel::parts::Ev3Gyroscope" for port "1###input######sensor1".`
Если есть какая-то информация, почему гироскоп у ev3 - скалярный, это было бы полезной информацией. Пока что пытаюсь сама понять.

iakov, 2019-07-26:
Draft in #408 with #419, #416, #411

closed 2019-07-26 by iakov