trikset/trik-studio #238: qrxc: not stable order of XML attributes

IKhonakhbeeva, 2019-05-31:
Магия с фиксированным elements.h немного не до конца работает

[image]

iakov, 2019-05-31:
Это Qt 5.12 MinGW 7.3  различия Debug & Release сборок?

IKhonakhbeeva, 2019-06-01:
Это различия того, что сейчас лежит в репо и результата после любой сборки Qt 5.12 mingw7.3 debug/release x86/x64
Порядок того, что получается - не фиксирован и гуляет от сборки к сборке

iakov, 2019-06-06:
Ну вот есть такая проблема, да. И это проблема Qt, потому что не поддерживают normalized XML. https://bugreports.qt.io/browse/QTBUG-47566

iakov, 2019-06-06:
Пишут в документации, что надо лишь 0 выставить, что мы и делаем перед вызовом qrxc. https://doc.qt.io/qt-5/qhash.html#algorithmic-complexity-attacks

iakov, 2019-06-16:
@IKhonakhbeeva как оно сейчас после #257  ? Стало лучше?

IKhonakhbeeva, 2019-06-18:
Больше подбной ситуации не возникало

closed 2019-06-21 by iakov