trikset/trik-studio #269: Vm program validation

anastasiia-kornilova, 2019-06-10:
"Ошибка на EV3: ""vm program validation"" 
test_paral - работает
test_paral_zachet - не работает
test_paral_zachet_correct - работает"

anastasiia-kornilova, 2019-06-10:
https://drive.google.com/open?id=1tb0j4tB07-eJqYPiQAs7JyYUO0qBA9ew -- здесь программы

AlekseiPrivalihin, 2019-09-16:
Возможно, проблема связана с некорректной генерацией завершающих меток и прыжков на них, а именно:
1) В файле trik-studio\plugins\robots\generators\ev3\ev3RbfGenerator\templates\threads\implementation.t вместо __programEnd метка называется programEnd (скорее всего, опечатка).
2) В файле trik-studio\plugins\robots\generators\generatorBase\src\gotoControlFlowGenerator.cpp в функции cloneFor при создании копии аргумент isThisDiagramMain всегда false, что приводит к генерации прыжка на метку __subprogramEnd вместо __programEnd. Частичным решением для EV3 была бы установка этого аргумента в зависимости от переданного функции cloneFor параметра cloneForNewDiagram (такая замена действительно генерирует код, который выглядит корректно, хотя тестирование того, исправляет ли это ошибку на живом EV3 проведено пока не было), однако, для других платформ, где блок объединения параллельных процессов, насколько я понимаю, позволяет создавать параллельные процессы в подпрограмме (как себя поведет такая конструкция в EV3 тоже надо проверить), такое решение вряд ли будет корректным.

AlekseiPrivalihin, 2019-09-16:
Кроме того, стоит проверить, что этот параметр нигде не используется в другом смысле (например, идентификация основного потока, а не диаграммы), чтобы изменение не испортило другие места в программе.

AlekseiPrivalihin, 2019-09-18:
Замена не убирает ошибку валидации для исходной программы, но убирает ошибку валидации для, например, такой программы:
[Illuminati.zip](https://github.com/trikset/trik-studio/files/3626544/Illuminati.zip)

iakov, 2019-10-29:
@AlekseiPrivalihin , так мы видим, где код некорректно сгенерился и понимаем, каким он должен быть, или вообще пока не ясно, почему нам EV3 выдаёт ошибку?