trikset/trik-studio #392: Не копируются блоки в режиме редактора с помощью горячих клавиш

shirokolobov, 2019-07-16:
Точнее, копируются, но после третьего нажатия ctrl+c

IKhonakhbeeva, 2019-07-22:
Не получается воспроизвести, ctrl+c ctrl+v работает постоянно корректно с первого раза.
@shirokolobov Регулярно повторяется или только при каких-то условиях?

shirokolobov, 2019-07-23:
@IKhonakhbeeva пробовал много раз - действительно сложно отловить. Вот когда точно не работает (на винде):
1. Перезагружаем комп.
2. Запускаем студию.
3. Открываем программу.
4. Кликаем сразу на блок.
5. Ctrl+C.
6. Ctrl+V. Не работает.

Если далее кликнуть в любое место на сцене, то всё работает. После перезапуска Студии тоже всё работает.

Еще бывает, что блок копируется на то же самое место и это можно не заметить (это надо кликнуть перед копированием в левый верхний квадрат блока). Но это к вышеописанному алгоритму это не относится.