trikset/trik-studio #461: Обработка строки "тело цикла"

anastasiia-kornilova, 2019-08-15:
"Прошу ввести обработку выражения ""тело цикла"" в вариации с несколькими пробелами и несколькими переносами строк. То есть принимать без ошибок выражения типа:
""тело
        
       цикла"".
Автоматически нормализовать их до одного переноса строки или одного пробела, если переносов нет.
Это нужно для лучшего визуального восприятия стрелки, поскольку сейчас она слишком растягивается."

iakov, 2019-08-26:
`QString::simplified`

IKhonakhbeeva, 2019-10-25:
Если такой вариант будет приниматься, вместо него все равно подставится "тело цикла"
Т.к. лейблы перерисовываются глядя на логику, а логика будет воспринимать это как внутреннее значение (iteration). Ну и в сейв сохраняется тоже внутреннее значение, поэтому лишние пробелы/переносы не сохранятся.

Возможные решения:
1) заменить "тело цикла" на что-то покороче
2) сделать всегда "тело'\n'цикла"
3) научиться ресайзить лейблы (сложность пока не проверялась)

iakov, 2019-10-26:
Пока откладываем.

anastasiia-kornilova, 2019-10-26:
Ну вообще там можно просто сглазить за свойством в репозиторий сделать simplified и сравнить с tr('body')

anastasiia-kornilova, 2019-10-26:
https://github.com/anastasiia-kornilova/trik-studio/tree/editor/461_draft -- вот здесь можно посмотреть версию с этим хаком, она текущую проблему решает, надо только дочистить. На мой взгляд, проблема эта более глобальная -- хочется так делать не только для "тело цикла", а чтобы симплифай строк был частью генератора, то есть автоматически подчищать лишние пробелы, переносы строк и пр.

anastasiia-kornilova, 2019-10-26:
Эта штука сохраняет правильное значение в сам сейв. Надо только адаптировать ее лейблы.

IKhonakhbeeva, 2019-10-29:
Да, действительно, этот хак работает
Но он очень "узкий", я как раз говорила про то, как глобально решить эту проблему.
Сохраняет в сейв он "правильное" значение, потому что оно (по мнению внутренней логики) неправильное.
https://github.com/trikset/trik-studio/blob/master/qrgui/editor/labels/label.cpp#L113
Постоянно тягается этот метод при взаимодействии с лейблами, чтобы текст предоставить в friendly-виде (т.е. логически "верное" значение, но в правильном переводе)

IKhonakhbeeva, 2019-10-29:
То есть можно сделать так, чтобы подчищались лишние пробелы, если с ними ввели, но выводиться будет по итогу без пробелов