trikset/trik-studio #517: Баг в реалистичной физике

NastenkaMeretskaya, 2019-09-16:
https://drive.google.com/open?id=1WZjQqcMOhqTNqvAMy55TSTdTSxp60gxk

NastenkaMeretskaya, 2019-09-16:
[qreal.log](https://github.com/trikset/trik-studio/files/3615751/qreal.log)
[actions.log](https://github.com/trikset/trik-studio/files/3615752/actions.log)

IKhonakhbeeva, 2019-09-24:
2. Надо более четкую последовательность действий для воспроизведения. Иногда мелькает, но пока что не понятно, как повторить.

NastenkaMeretskaya, 2019-09-24:
>    1. Надо более четкую последовательность действий для воспроизведения.

Ага, спасибо!

NastenkaMeretskaya, 2019-09-24:
А, я тупняк.
Нужно описать, как повторить?

вт, 24 сент. 2019 г., 20:40 Anastasiya Meretskaya <meretskaya@trikset.com>:

IKhonakhbeeva, 2019-09-24:
Да, а то очень рандомный (вроде) баг, не получается его отловить нормально

PR #607 (2019-11-02): Drawing fixes
Fixed #600 
Fixed #517
closed 2019-11-02 by mergify[bot]