trikset/trik-studio #519: Annoying naming schema for EV3

iakov, 2019-09-16:
Если название файла оканчивается на цифру, при формировании байткода возникает дополнительная сквозная нумерация. Ее надо убрать, слишко много файлов загрузится на контроллер.
Например, имя файла - f02.qrs. Формируем байткод f02.lms и загружаем его на EV3. Все нормально. Затем вносим изменения в диаграмму, закрываем сформированное окно байткода и снова генерируем его. Получается f021.lms. Затем f022.lms. И так далее.

IKhonakhbeeva, 2019-10-24:
Новый номер будет только если сформированный код изменяли (и сохраняли). Иначе заменится предыдущий сохранённый (с, возможно, важными ручными изменениями) вариант.

Предложение: если хочется сохранить изменённую lms, всегда надо выдавать окно сохранения в новое место, а не автоматически сохранять в appdata (где оно изначально лежит после создания)
Или при кодогенерации спрашивать, заменить ли старый вариант.

В общем, надо обсуждать поведение

iakov, 2019-10-24:
Ок, пока признаем "не ошибкой": не надо пытаться править код ручками.

closed 2019-10-24 by iakov