trikset/trik-studio #532: Incorrect gyrosensor behaviour

iakov, 2019-09-22:
"Гиродатчик возвращает неадекватные значения. В первом примере движения по квадрату в режиме 2D-модели после первого поворота крутится на месте, а на реальном роботе выполняет повороты на 30-50 градусов.
Во втором примере реализовано ожидание в цикле
 While. Условие sensor1<a выдает ошибку ""Несоответствие типов"".

[drive-download-20190922T102612Z-001.zip](https://github.com/trikset/trik-studio/files/3639611/drive-download-20190922T102612Z-001.zip)

IKhonakhbeeva, 2019-09-22:
Первый пример:

Такое поведение связано с тем, что там на втором круге идет проверка, что гиродатчик должен быть >=180
А у нас показания от -180 до 180 (то есть берем по модулю)
Сам гиродатчик ведет себя адекватно, просто надо обсуждать, что мы хотим тогда с гироскопа получать (насколько я помню, как раз было озвучено про -180..180)

У меня на реальном роботе повороты на ~90 градусов, никаких 30-50 не вижу. Но это может быть обусловлено починкой переключения физики. Надо попросить проверить этот пункт на новой версии.

IKhonakhbeeva, 2019-09-22:
Второй пример:
Быстрое решение - сказать использовать sensor1[0], так как гироскоп у нас возвращает данные массивом.

IKhonakhbeeva, 2019-10-03:
Проблема с реальным роботом выделена в отдельный ишью #551

PR #550 (2019-10-04): Gyroscope fixes
Fixed #532 
Добавлен варнинг для #420
closed 2019-10-04 by mergify[bot]