trikset/trik-studio #551: EV3 gyro on a real robot does not work correctly

IKhonakhbeeva, 2019-10-03:
Выделен из #532 

Гиродатчик возвращает неадекватные значения. В первом примере движения по квадрату на реальном роботе выполняет повороты на 30-50 градусов.

[drive-download-20190922T102612Z-001.zip](https://github.com/trikset/trik-studio/files/3639611/drive-download-20190922T102612Z-001.zip)

iakov, 2019-10-04:
Это просто баг, и он в 2019.7. Не надо ставить postponed просто так на мелкие баги.

IKhonakhbeeva, 2019-10-23:
Не работал вывод данных по sensor1 и sensor1[0]. Может, ориентировались на эти показания? В #581 корректно работает sensor1[0] на вывод. Если поворачивать робота руками (вместо моторов) на первом примере, то отрабатывает как и должен, на каждые 90 градусов. Так что, видимо, not-reproducible.

IKhonakhbeeva, 2019-10-24:
Частично (в проекте был вывод на дисплей значений с гироскопа). Но основная проблема именно в waitForGyro
Кодогенерация ожидания гироскопа не изменилась (ни при переходе на "векторный" гироскоп, ни в 581), так что это может быть связано еще с чем-то

IKhonakhbeeva, 2019-10-24:
То есть и до #581 эта программа отрабатывала корректно на поворот, но неправильно только на вывод sensor1