trikset/trik-studio #556: Сохранение моторов в модели мира при повторном открытии файла

NastenkaMeretskaya, 2019-10-04:
Не сохраняется новая конфигурация моторов в студии при повторном открытии. Моторы возвращаются в состояние "по умолчанию".

[actions.log](https://github.com/trikset/trik-studio/files/3690220/actions.log)
[qreal.log](https://github.com/trikset/trik-studio/files/3690221/qreal.log)

anastasiia-kornilova, 2019-10-07:
Сейчас конфигурация сохраняется в qrs. То есть при повторном открытии проекта конфигурация сохраняется. Если открывать НОВЫЙ проект открывается дефолтная конфигурация с M3, M4. Что мы хотим сохранять? Последнюю конфигурацию для использования в НОВОМ проекте? @iakov , @NastenkaMeretskaya

NastenkaMeretskaya, 2019-10-07:
Проблема в том, что конфигурация не всегда сохраняется при повторном
открытии файла.

anastasiia-kornilova, 2019-10-07:
@NastenkaMeretskaya , а можно саму qrs-ку?

NastenkaMeretskaya, 2019-10-07:
В некоторых qrs через несколько повторных сохранений проблема уходит, в
этом примере не ушла.

пн, 7 окт. 2019 г. в 12:28, Anastasiya Meretskaya <meretskaya@trikset.com>:

>
>
> пн, 7 окт. 2019 г. в 12:17, Anastasiia Kornilova <notifications@github.com
> >:
>
>> @NastenkaMeretskaya <https://github.com/NastenkaMeretskaya> , а можно
>> саму qrs-ку?
>>
>>
>
>
>
> *С уважением,*
>
> *Мерецкая Анастасия*
> *методист ООО "КиберТех"*
> *+7 (981) 773-92-18*

iakov, 2019-10-24:
> Сейчас конфигурация сохраняется в qrs. То есть при повторном открытии проекта конфигурация сохраняется. Если открывать НОВЫЙ проект открывается дефолтная конфигурация с M3, M4. Что мы хотим сохранять? Последнюю конфигурацию для использования в НОВОМ проекте? @iakov , @NastenkaMeretskaya

В новом проекте пусть будет по умолчанию. Можем завести отдельный тикет на "При создании нового проекта желательно переиспользовать конфигурацию робота из текущего открытого", если это удобно было бы.

NastenkaMeretskaya, 2019-10-24:
Я проверю мастер. Проблема в том, что в task не сохранилась конфигурация
при повторном открытии.

iakov, 2019-11-22:
Закрываю, поскольку дополнительной информации так и не получил.

closed 2019-11-22 by iakov