trikset/trik-studio #579: Проблема с фоновым изображением (не подгружается)

iakov, 2019-10-23:
> "При загрузке модели мира фоновое изображение не подгружается, хотя лежит в той же папке."

IKhonakhbeeva, 2019-10-25:
А разве должно?
Фон - такая же картинка, его надо либо сохранять в сейв, либо искать по указанному при создании пути (а не "в той же папке")
Возможно, недопонимание из-за того, что у в "настройки" фона сложнее зайти и непонятно, что там тоже есть галочка "сохранять в сейв".

Предложение: фон сделать автоматически сохраняемым в сейв (сейчас по умолчанию сохраняется только путь до картинки)

iakov, 2019-10-26:
Давай всегда картинки (и фон) по умолчанию сохранять. Пусть галочка/опция будет включена по умолчанию. А то нам будут присылать сейвы с программами, в которых картинок нет.

PR #588 (2019-10-26): Save image items to qrs by default
Fixed #579
closed 2019-10-26 by mergify[bot]