trikset/trik-studio #592: Incorrect main window rendering win scaled fonts

iakov, 2019-10-28:
> У одного из педагогов на ноуте стоят увеличенные шрифты - 150%. При запуске студии не видно левую панель (переключение режимов редактор/отладка). Решилось только установкой шрифтов 100%. (Думаю, и в прошлых версиях эта бага есть, просто не сталкивались раньше.)

khodand, 2019-11-19:
Стоят увеличенные именно шрифты? У меня на ноуте интерфейс 150% и все нормально

iakov, 2019-11-22:
Наверное, только шрифты. Попробуй, если есть такая опция в Виндоуз. @shirokolobov , не помнишь детали?

shirokolobov, 2019-11-22:
> Наверное, только шрифты. Попробуй, если есть такая опция в Виндоуз. @shirokolobov , не помнишь детали?

Не очень. Винда, старый ноут с экраном 4:3.

iakov, 2019-11-27:
1024х768 действительно может подглючивать, да ещё и с увеличенными шрифтами. Ладно пока снимаю майлстоун, если будет многих мучить, будем думать.