trikset/trik-studio #596: Gyrosensor problems with EV3

iakov, 2019-10-29:
На 2019_6-134-g310547 (это уже с последними фиксами #581, но мб юзеры "ошибаются", надо проверить.

> "1. На EV3 возникает ошибка при обращении к гиродатчику.
> 2. В режиме 2D диапазон измерений при повороте моторами адекватен (без ограничений), но при вращении мышкой показывает странное, например, от -315 до 45."

[gyro_bug.zip](https://github.com/trikset/trik-studio/files/3785291/gyro_bug.zip)

IKhonakhbeeva, 2019-10-29:
1. Ошибка при обращении именно способом sensor1[0] ?

IKhonakhbeeva, 2019-10-29:
1. Посмотрела генерацию, вроде должно нормально работать. Проверю на ev3 в пятницу.
2. Интересная проблема.. Проблема в какой-то степени в том, что "как определять, в какую сторону крутили мышкой"? Посмотрю, что с этим можно придумать.

IKhonakhbeeva, 2019-11-01:
1. Какая ошибка возникает? У меня все работает, как в режиме по юзб на реального робота, так и 2d модель

PR #606 (2019-11-01): Fix gyro angle in case of rotation with mouse
Fixed #596
closed 2019-11-01 by mergify[bot]