trikset/trik-studio #626: TS silently ignores large images

iakov, 2019-11-07:
> Не загружаются файлы изображений ни как фон, ни как рисунок.
Это поведение при больших картинках. Надо просто как-то предупреждать, лучше даже ругаться с ошибкой или выдавать диалоговое окно с ошибкой.

IKhonakhbeeva, 2019-11-08:
У меня png в 20 МБ в качестве 4К вполне загружает и как фон, и как рисунок, с предупреждением, что это может заморозить немного процесс выполнения.
Можно сюда скинуть конкретную картинку (или больше подробностей), с которой не работает, может быть?

iakov, 2019-11-09:
Запросил. Ждём.

iakov, 2019-11-11:
Вот файл [image]

PR #641 (2019-11-19): Fix JPEG images loading
Fixed #626
closed 2019-11-19 by IKhonakhbeeva