trikset/trik-studio #646: Incorrect condition gereration

iakov, 2019-11-20:
> Условие. Многосоставное.
> Вторая проблема, которую не решили, возникла с задачей подсчета перекрестков с помощью двух датчиков освещенности. Двигаемся по линии и считаем перекрестки.
Такое ощущение, что реальный робот по факту выполняет последовательно верхнюю и нижнюю ветку условия. По какой причине? Не правильное вычисление логического выражения. Если писать просто условие без and или && оно срабатывает. А когда условие многосоставное, как в задаче, результат выражение это значение одной из переменных grey1 или grey2.

AlekseiPrivalihin, 2019-11-22:
извлеченный из архива файл не открывается ТРИК студией, причем ни 2019.6, ни текущей версией из мастера

iakov, 2019-12-06:
Вполне себе открывается. Мб кто-то забыл сделать `git pull upstream`?