trikset/trik-studio #678: TS crashes when incorrect js-constructions are used

anastasiia-kornilova, 2019-12-07:
Examples:

```
var Cell_type =
{
	UNKNOWN			: -1,
	WALL			: Number.MAX_SAFE_INTEGER, 	
	EMPTY			: 0
};
```
Crash due to Number.MAX_SAFE_INTEGER

```
readings = []
readings += smth
maxReading = Math.max(readings)
```
Crash due to incorrect syntax in max for arrays

It is better to have some error message for user about incorrect behaviour

iakov, 2019-12-10:
@anastasiia-kornilova , how to reproduce the issue in master branch?

Program 
```readings = [1,2]
maxReading = Math.max(readings)
print(maxReading)```
outputs `NaN`. Seems fine.

anastasiia-kornilova, 2019-12-10:
Hmm... Maybe the problem is in more specific actions. Attached file.
[example.tar.gz](https://github.com/trikset/trik-studio/files/3946740/example.tar.gz)

anastasiia-kornilova, 2019-12-29:
Here is qrs with first problem reproduced on the current master (2019.8)
[cell_type.qrs.zip](https://github.com/trikset/trik-studio/files/4008536/cell_type.qrs.zip)