trikset/trik-studio #679: 2D model crashes in checker when script execution is stopped

anastasiia-kornilova, 2019-12-07:
gdb trace:

```
Core was generated by `./bin/2D-model --platform minimal -b ./main-copy.qrs --report /trikStudio-check'.
Program terminated with signal SIGABRT, Aborted.
#0  0x00007f9380589428 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
[Current thread is 1 (Thread 0x7f93790ba780 (LWP 44))]
(gdb) bt
#0  0x00007f9380589428 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007f938058b02a in __GI_abort () at abort.c:89
#2  0x00007f9380f9cefc in QMessageLogger::fatal(char const*, ...) const () from /trikStudio-checker/bin/libQt5Core.so.5
#3  0x00007f9380fbac1c in QThread::~QThread() ()
   from /trikStudio-checker/bin/libQt5Core.so.5
#4  0x00007f9376d2df27 in trikScriptRunner::TrikJavaScriptRunner::~TrikJavaScriptRunner() ()
   from /trikStudio-checker/bin/libtrikScriptRunner.so.1
#5  0x00007f9376d2df79 in trikScriptRunner::TrikJavaScriptRunner::~TrikJavaScriptRunner() ()
   from /trikStudio-checker/bin/libtrikScriptRunner.so.1
#6  0x00007f9376d29c05 in trikScriptRunner::TrikScriptRunner::~TrikScriptRunner() () from /trikStudio-checker/bin/libtrikScriptRunner.so.1
#7  0x00007f9376e89a43 in trik::TrikTextualInterpreter::~TrikTextualInterpreter() ()
   from /trikStudio-checker/bin/librobots-trik-kit-interpreter-common.so.1
#8  0x00007f9376e89a69 in trik::TrikTextualInterpreter::~TrikTextualInterpreter() ()
   from /trikStudio-checker/bin/librobots-trik-kit-interpreter-common.so.1
#9  0x00007f9376e767d3 in trik::TrikKitInterpreterPluginBase::~TrikKitInterpreterPluginBase() ()
   from /trikStudio-checker/bin/librobots-trik-kit-interpreter-common.so.1
#10 0x00007f937411d5ac in trik::TrikV62KitInterpreterPlugin::~TrikV62KitInterpreterPlugin() ()
---Type <return> to continue, or q <return> to quit---
   from /trikStudio-checker/bin/plugins/tools/kitPlugins/librobots-trik-v62-interpreter.so
#11 0x00007f938117ff49 in ?? ()
   from /trikStudio-checker/bin/libQt5Core.so.5
#12 0x00007f93811738b7 in QPluginLoader::unload() ()
   from /trikStudio-checker/bin/libQt5Core.so.5
#13 0x00007f9382664e2c in qReal::details::PluginManagerImplementation::~PluginManagerImplementation() ()
   from /trikStudio-checker/bin/libqrutils.so.1
#14 0x00007f93821b612d in interpreterCore::KitPluginManager::~KitPluginManager() ()
   from /trikStudio-checker/bin/librobots-interpreter-core.so.1
#15 0x00007f938217824e in interpreterCore::RobotsPluginFacade::~RobotsPluginFacade() ()
   from /trikStudio-checker/bin/librobots-interpreter-core.so.1
#16 0x0000000000409f80 in ?? ()
#17 0x0000000000407d51 in ?? ()
#18 0x00007f9380574830 in __libc_start_main (main=0x4072f0, argc=13, 
    argv=0x7fff3036eea8, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fff3036ee98)
    at ../csu/libc-start.c:291
#19 0x00000000004082f9 in ?? ()

anastasiia-kornilova, 2019-12-07:
Looks like here: https://github.com/trikset/trikRuntime/blob/87cc37a9f06a86da5966012ea6f4de3fdf102111/trikScriptRunner/src/trikJavaScriptRunner.cpp#L57 thread isn't finished even after `wait`, but we destroy it when coming out from the scope.

As temporary solution maybe it is better to terminate thread, like:

```
if (!mWorkerThread.wait(1000)) {
  mWorkerThread.terminate();
  mWorkerThread.wait();
}

iakov, 2019-12-14:
Please, check the proposed solution locally, if you already know how to reproduce the problem.

iakov, 2020-10-09:
It is fixed for 2020.6, isn't it?

closed 2020-11-16 by iakov