trikset/trik-studio #724: Stop 2D model by pressing F5

anastasiia-kornilova, 2020-01-28:
Now only running support under F5 button

IKhonakhbeeva, 2020-01-29:
Везде, где можно запустить по F5 (или Ctrl+F5 для трика), можно остановить по Shift+F5 (аналогично Shift+Ctrl+F5)

Единственное, что это криво работает, когда запускается код, а не блоки. Это можно поправить на Shift+F5, чтобы аналогично работало

anastasiia-kornilova, 2020-01-29:
Что ты имеешь в виду под "везде"?
Анивей, нужна хоть какая-то вариация. Если она есть, укажи, плиз, как она работает в студии для скриптов.

IKhonakhbeeva, 2020-01-29:
Для отправки на ev3 работает F5 (+Shift для остановки) и для блоков, и для скриптов
Для отправки на триковского робота (в режиме js и python) работает Ctrl+F5(+Shift для остановки) опять же и для блоков, и для скриптов
Для 2d-режима блочная диаграмма работает на F5(+Shift для остановки) и в режиме Edit, и Debug

Проблема только когда в 2d модели запускаем триковские скрипты. Если открыт Edit, то вообще ни запустить, ни остановить нельзя. Если при этом открыта сама моделька, то сейчас действительно можно только запускать (но локально уже добавила остановку на Shift+F5). На сами скрипты тоже, наверное, попытаться это повесить? Чтобы когда открыт скрипт, его можно было в 2d модели запустить?

anastasiia-kornilova, 2020-01-29:
Спасибо за развернутый ответ, поняла.

> На сами скрипты тоже, наверное, попытаться это повесить? -- Ага, если это не займет много времени, чтобы не пользователю не делать лишних движений. В любом случае, остановку можешь выкатить сейчас, остальное потом будем приукрашивать.

А еще я не поняла общую логику -- почему где-то без Ctrl, а где-то с ним, причем в модель -- "на роботов отправляем с Ctrl, а локально запускаем в модели без" это не укладывается. Надо либо понять, либо как-то унифицировать.

IKhonakhbeeva, 2020-01-29:
А вот почему где-то с Ctrl - понятия не имею. Так как никаких событий на просто F5 не повешено, то есть конфликта нет. Может, когда-то давно была глубокая мысль...

PR #727 (2020-02-02): Run and stop robot with F5 and Shift+F5 in code interpretation
Fixed #724
closed 2020-02-02 by mergify[bot]