trikset/trik-studio #761: Add ability to extend TRIK display in 2d model

anastasiia-kornilova, 2020-02-23:
Now it is very small to observe computer vision artifacts, like lines, points and etc.
[image]

iakov, 2020-02-27:
Идея от @anastasiia-kornilova 

> Зума будет недостаточно. Это очень маленькое увеличение получается. А сейчас просто глаза ломаешь. Я думаю, что надо что-то вроде:
> Кликнули на экран контроллера, экран заскейлился до размеров контроллера, кликнули опять -- уменьшился обратно. Это будет эффективнее. Но я не уверена, что на hdpi это спасет. Но всяко будет лучше.

iakov, 2022-06-03:
Есть задачи, где только рисуют на экране, либо рисование -- значительная часть занятия.