trikset/trik-studio #764: Вместе с подпрограммами экспортируется также "модель мира".

BorisADrozdov, 2020-02-26:
Вместе с подпрограммами экспортируется также "модель мира". 
Открывается если открыть файл с подпрограммами как проект.

iakov, 2020-02-26:
Если не будет импортироваться и затирать уже нарисованное, то пусть себе сохраняется.

BorisADrozdov, 2020-02-27:
Не то чтобы оно импортировалось и затирало. 
Модель мира просто обнуляется вместе с картой подключений тележки.

И мне кажется. что обнуляется и-за того, что пытается импортировться.

Я думаю, что карта подключений "тележки" НЕ ДОЛЖНА быть в "картине мира" (хоть и находится на той-же вкладке) - она должна быть связана с основной программой т.к. в основная программа привязана к тележке, а не карта мира.
Тогда любого робота можно будет поместить на любой полигон.

Т.е. правильное решение только одно: карту подключений тележки привязать и основной прогр. и убрать из "карты мира" (если она там есть).

Второе правильное решение - убрать "модель мира" и "карту подключений" из импорта/экспорта подпрограмм - и вопрос решится.

Альтернативное предложение - оставить портирование подпрограмм только через "коллекцию", НО тогда надо сделать Сохранение "Коллекции" в файл чтобы эти коллекции могли быть разными.

BorisADrozdov, 2020-03-01:
Через "коллекцию" не получится.
В "коллекции" с каждой подпрограммой так же любовно сохраняется "кара мира" и так же затирается текущая "кара мира" при загрузке подпрограмм из "коллекции".

PR #918 (2020-05-17): Fix world loading with subprograms
Fixed #773
Fixed #764
closed 2020-05-17 by iakov