trikset/trik-studio #768: Old version showed

BorisADrozdov, 2020-02-26:
При установке:
- создается ярлык с названием старой версии
- на титульной странице (запуск с ярлыка) - старая версия.

iakov, 2020-02-26:
Такое ощущение, что при установке не удалилась старая версия. Сбоев при установке не было?

BorisADrozdov, 2020-02-27:
Удалял начисто, затем устанавливал через инсталлер, через скрипт - всё едино.
И уж точно это никак не влияет на титульную страницу.

anastasiia-kornilova, 2020-02-27:
@BorisADrozdov, проверьте, куда идёт ярлык, точно ли в новую студию?

BorisADrozdov, 2020-02-28:
Естественно. Могу всё проверить еще раз.

BorisADrozdov, 2020-02-28:
[image]
[image]
Это после "чистой" установки - всё удалил через установку/удаление, установил через инсталлер, т.е. ошибка с ярлыком - в инсталлере. С титульной страницей - видимо, в титульной странице :-)

anastasiia-kornilova, 2020-02-28:
Хм, у версии грязный хвост, откуда её берете?

anastasiia-kornilova, 2020-02-28:
В общем, то что скидывали в слак и не должно было быть с номером, дали просто текущий срез для проверки функциональности. Версия обновляется тогда, когда её затегировали - заморозили. https://dl.trikset.com/ts/fresh/trik-studio-installer_2020.1-i686.exe - вот эта, кажется, затеганная, попробуйте её

BorisADrozdov, 2020-02-28:
Да, та, что Яков в Слаке скинул. https://dl.trikset.com/ts/fresh/trik-studio-installer-master-i686-auto.zip
Пробую новую.

BorisADrozdov, 2020-02-28:
Ярлык - Ок, 
Титульная - Ок
Тикет можно "закрывать".

closed 2020-02-28 by anastasiia-kornilova