trikset/trik-studio #770: Зависимость траектории от течения времени

IKhonakhbeeva, 2020-02-26:
траектория движения тележки зависит от "скорости" времени. Передвижения простейшие, контроль по 1 энкодеру. При 100% движется наиболее аккуратно. При 200-300-400% иногда не доезжает до конца лабиринта - упирается в стенку.

iakov, 2020-03-01:
А разве это не дубликат #401 ?

BorisADrozdov, 2020-03-01:
Как это по причинам ("изнутри") - не знаю. Может, причины одни.
Снаружи - не то-же самое.
Об энкодерах - ни слова.
Траектория косеет при изменении временного масштаба. 
О чём говорит? 
Скорее всего, о том, что итерации, видимо, идут по времени и интервал не изменяется с изменением временного масштаба. Сделать пропорциональным или привязать к "внутреннему ввремени" - и не будет "косеть".
Мне так кажется ... издалека ...