trikset/trik-studio #778: Is it possible to import thirdparty libraries like numpy, scipy, etc?

anastasiia-kornilova, 2020-03-02:
If yes, what steps should be done?
If no, it would be a good feature to support.

khodand, 2020-03-06:
It's possible, a user just has to put his package to the script folder. That will work fine.

anastasiia-kornilova, 2020-03-06:
Wow! Really?

khodand, 2020-03-06:
I've checked that in Studio, but there is no reason why it won't work for the robot. The question for me is, can we deploy whole directories from Studio to the controller?

anastasiia-kornilova, 2020-03-06:
Deploy is not good way -- python should be compiled for arm. It seems to me, that we have some package manager on the controller to simply install. 

@khodand , in current TS import for files in the same directory doesn't work. You mean, that it works with your last fix?

anastasiia-kornilova, 2020-03-06:
About putting library in the same directory -- I'm a little bit confused. Probably it is better to extend system.path in the code with path to libraries.

iakov, 2020-03-10:
_AS IS_ disclamer: this advice has no warranty.
To use popular libraries with TS one can use separate standalone Python environment (local, system-wide, or cona/venv managed) to install all the desired libraries. The Python version *must* be the same as one used in TS. (Python3.7 for TS2020.1).  TRIK_PYTHONPATH must be set correctly for `trik-studio` process. Use `python -c 'import sys; import os; print(os.pathsep.join(sys.path))'` from the environment to get proper value for TRIK_PYTHONPATH. Should work for all platforms.