trikset/trik-studio #783: Incorrect reaction on saving action in close dialog

anastasiia-kornilova, 2020-03-04:
При закрытии приложения, вылазиет окно с вопросом "Сохранить текущий проект?". При нажатии на кнопку "отмена" Trik Studio закрывает твою программу, не сохраняя её.

IKhonakhbeeva, 2020-03-04:
Это когда с js работаешь?

anastasiia-kornilova, 2020-03-04:
"Почти всё то, что вы перечислили. 
Если открыт любой js-файл, то при попытке завершения Trik Studio, (хоть нажатием на на кнопку закрытия в правом верхнем углу), появляется окно с вопросом "Сохранить текущий проект?".
В диалоговом окне есть три кнопки: "Сохранить", "Не сохранять", "Отмена".
При нажатии на кнопку "Отмена" Trik Studio закрывает js-файл, не сохраняя его. "

BorisADrozdov, 2020-03-07:
Как понимаю, ТС вообще все вторичные окна (окромя основного) закрывает (подпрограммы, ЯС, ...) при подобной процедуре. Т.е. надо не закрывать все окна перед выдачей запроса. Или "сечь" изменения в окне ЯС.

PR #792 (2020-03-08): Fix saving no-qrs tabs
Fixed #783
closed 2020-03-08 by iakov