trikset/trik-studio #803: Diagram rendering problem

iakov, 2020-03-11:
Проблема с порядком/прозрачностью на сцене

https://www.youtube.com/watch?v=pN0Posq804k

IKhonakhbeeva, 2020-03-24:
Надо обсуждать, чего именно хочется. Т.к. лейбл - потомок блока, они над каждым другим блоком вместе либо выше, либо ниже, т.е нельзя, например, сделать так, что все лейблы положить "чуть ниже", чем все блоки (на данный момент вне селекта порядок определяется порядком добавления - и не вижу способов отличить, какой блок хочется видеть ниже или выше)

IKhonakhbeeva, 2020-03-24:
Или для каждого блока настроить прозрачный там фон или белый, а заодно и дефолт поменять на будущее (ушла куча времени, чтобы понять, где и как эта прозрачность вообще настраивается)
[для каждого элемента вот тут](https://github.com/trikset/trik-studio/blob/master/plugins/robots/editor/common/robotsMetamodel.xml#L170)
[дефолт тут](https://github.com/trikset/trik-studio/blob/master/qrxc/label.cpp#L47)