trikset/trik-studio #804: Strange draing artifacts when moving objects

iakov, 2020-03-11:
https://www.youtube.com/watch?v=6O6AAahT5Ds

BorisADrozdov, 2020-03-11:
2020.1
Сегодня столкнулся с данным эффектом.
Последствия:
- "тележка" так и ушла в "параллельное измерение" и не появлялась после всяких манипуляций.
- датчики продолжали двигаться как ни в чём ни бывало.
Главное:
- Сохраненная после данного инцидента "карта мира" стала приводить к крашу ТС при загрузке.

[3.6 Массивы_тупики отдельно 3v5.zip](https://github.com/trikset/trik-studio/files/4321129/3.6._.3v5.zip)

IKhonakhbeeva, 2020-03-13:
Последний пункт - все та же проблема со стенами, здесь она никак не влияет
К слову, проблема с крашем уже починена, можно в этом убедиться, скачав последнюю версию студии

PR #806 (2020-03-14): Fixed robot disappearance
Fixed #804
closed 2020-03-14 by iakov