trikset/trik-studio #839: Крэш TRIK Studio 2020.1-NTI1-38-g971c0e-dirty , связанный с датчиками.

BorisADrozdov, 2020-04-08:
Загружаю проект - в 2D всё работает.
Жму "сгенерировать код". Обе ТС пишут:
"[19:54:58]: ОШИБКА: Сенсор на порту A5 не сконфигурирован. Выберите его тип на панели "Настройки сенсоров" справа." (хотя всё подключено/сконфигурировано)
Подобное происходит довольно часто, рецепт - переподключить датчики.
Переподключаю. 
Далее 2020.1 (от 29 февраля) генерит код,
2020.1-NTI1-38-g971c0e-dirty - вываливается

Имя события проблемы:	APPCRASH
  Имя приложения:	trik-studio.exe
  Версия приложения:	0.0.0.0
  Отметка времени приложения:	5e8d7b94
  Имя модуля с ошибкой:	msvcrt.dll
  Версия модуля с ошибкой:	7.0.7600.16385
  Отметка времени модуля с ошибкой:	4a5bda6f
  Код исключения:	c0000005
  Смещение исключения:	0001e631
  Версия ОС:	6.1.7601.2.1.0.256.48
  Код языка:	1049
  Дополнительные сведения 1:	0a9e
  Дополнительные сведения 2:	0a9e372d3b4ad19135b953a78882e789
  Дополнительные сведения 3:	0a9e
  Дополнительные сведения 4:	0a9e372d3b4ad19135b953a78882e789

[2.1.11 Кучугурный(зелёный красный)01.zip](https://github.com/trikset/trik-studio/files/4451787/2.1.11.01.zip)

iakov, 2020-04-08:
-dirty в имени версии и креш -- это в #840  
Осталась только глючащая привязка датчика

iakov, 2020-06-26:
The rest looks like #931  or #802

closed 2020-06-26 by iakov