trikset/trik-studio #901: Robot continues rotating even if motors are stopped.

iakov, 2020-05-04:
> "Остановка моторов не останавливает робота, а продолжает его вращение с небольшой скоростью. В примере робот движется по лабиринту. После остановки в каждой подпрограмме добавлены задержки 3 с, во время которых он вращается.  Эта ошибка проявляется не сразу. Иногда надо поработать со средой, чтобы после очередного запуска робота начало вращать"

IKhonakhbeeva, 2020-05-04:
очень интересное поведение - первое впечатление это какое-то "накопление" дельты угла, на который роботу надо повернуться. Если несколько раз позапускать эту программу, то потом робот даже на обычном "ждать 10 секунд" медленно вращается сам по себе

iakov, 2020-05-04:
Да, вращается.

PR #939 (2020-05-28): Fix robot post-rotation
Fix #901