trikset/trik-studio #922: "open" support in checker

anastasiia-kornilova, 2020-05-17:
For now, the only way to use input in python with checker is `script.readAll`. Probably, for checker we need to redefine "open" to make standard students' programs compatible with checker.

DCNick3, 2020-06-15:
I think that a better solution would be to use stdin for such purposes. It's a common solution in competitive programming.

This way one would still be able to use file APIs (I used those for storing calibration values, for example). Also this would remove the need to specify a specific filename (which will be ignored).

However, if one would want an interoperability between checker, trik studio and real robot (is it really needed that much?), stdin should be redirected there as well. It's not clear what information should be supplied in these cases.

Another possible solution may be an introduction of new API for getting test input. Something like `script.getTestInput()`.

hircumg, 2020-06-22:
@DCNick3 

By the way, this issue was initiated during robotics competition while participants used `open` insted of `script.readAll`. 
Hence they have been failed during testing. Probalby, because they allready known python and also I belive first link in google provides such file reading.

DCNick3, 2020-06-22:
Well, that can be explained in the competition rules. If "stdin" route would be taken, it will work flawlessly: this is the most standard way, stackoverflow is full of examples of reading from there. If `script.getTestInput()` way would be taken than the input is no longer a file, and so the misunderstanding is gone: there is a special API to get test input, it doesn't have anything to do with files.