trikset/trik-studio #986: Implement user-friendly Python engine error reporting

iakov, 2020-06-25:
As of 2020.3 it looks ugly: single line string from Python interpreter with incorrect source connection.

ikonovalova, 2020-06-30:
Пример 1. Ошибка в строке 22. На скриншоте выводимая ошибка
[image]
``` import sys
import time
import random
import math

class Program():
  __interpretation_started_timestamp__ = time.time() * 1000

  pi = 3.141592653589793

  def execMain(self):
    brick.motor("M3").setPower(50)
    brick.motor("M4").setPower(100)
    
    script.wait(3000)
    
    brick.motor("M3").powerOff()
    brick.motor("M4").powerOff()
    
    brick.marker().down(red)
    
    brick.motor("M3").setPower(100)
    
    script.wait(1000)
    
    print(brick.battery())
    brick.led().orange()
    brick.say("Привет, я ТРИК")
    
    brick.motor("M3").setPower(-(100))
    brick.motor("M4").setPower(-(100))
    
    script.wait(3000)
    
    brick.stop()
    return

def main():
  program = Program()
  program.execMain()

if __name__ == '__main__':
  main()
```

ikonovalova, 2020-06-30:
Пример 2. Ошибка в строке 17 (True). 
[image] 
```python {.line-numbers} {highlight=17}
import sys
import time
import random
import math

class Program():
  __interpretation_started_timestamp__ = time.time() * 1000

  pi = 3.141592653589793

  def execMain(self):

    
    brick.motor("M3").setPower(100)
    brick.motor("M4").setPower(100)
    
    while true:
      if (brick.sensor("A1").read() < 50):
        brick.display().showImage("media/trik_smile_sad.png")
      else:
        brick.display().showImage("media/trik_smile_normal.png")
      script.wait(50)
      

def main():
  program = Program()
  program.execMain()

if __name__ == '__main__':
  main()
```

PR #1398 (2021-08-02): Make error labels multi-line
[image]
Fix #986
closed 2021-08-02 by iakov