trikset/trik-studio #1026: Проблема с установкой TRIK-Studio на macOs Big Sur.

daneel-lednikov, 2020-07-28:
Устанваливаю trik-studio на Big Sur. Такая ошибка:
`Ошибка во время процесса установки (ru.qreal.root): Не удалось исполнить (неожиданный код завершения: 6): «/Users/lednikov/TRIKStudio/TRIK Studio.app/Contents/MacOS/trik-studio --clear-conf»`

При нажатии `повторить` все тоже самое, если пропускаю - устанавливается, но не запускается. 
ОС: Big Sur Beta 20A4300b

iakov, 2020-07-28:
Проверьте, пожалуйста, https://dl.trikset.com/ts/fresh/installer/trik-studio-installer-mac-master.dmg

daneel-lednikov, 2020-07-29:
Установилось успешно. Запускаю, в доке появляется значок и на этом все. Окна программы нету, не запускается.

iakov, 2020-08-25:
@daneel-lednikov , попробуйте, пожалуйста, свежую сборку https://dl.trikset.com/ts/fresh/installer/trik-studio-installer-mac-master.dmg 
Лучше из терминала запустить `"$HOME/TRIKStudio/TRIK Studio.app/Contents/MacOS/trik-studio"`. Интересно, что будет выведено в окно терминала.

iakov, 2021-01-03:
Fixed for 2020.6

closed 2021-01-03 by iakov