trikset/trik-studio #1160: silent install paramteters

serzhenko, 2020-11-08:
Добавьте, пожалуйста, поддержку «тихой» установки с помощью ключей командной строки для того, чтобы можно было автоматизированно развёртывать TRIK Studio в учебных лабораториях

iakov, 2020-11-16:
Такая возможность есть. Вы обращались в поддержку и Вам не ответили? 

Давайте тогда вместе подумаем, что именно хочется.

Вот скрипт автоматизации установки на Windows: https://github.com/trikset/trik-studio/blob/master/installer/reinstall-trik-studio.cmd , он использует скрипты удаления и установки, они в той же папке.

Там же рядом есть пример для мака, пример для линукса есть в скриптах автоматизации тестирования. В общем, главное там -- вызвать инсталлятор с правильными опциями и подсунуть специальный скрипт инсталлятору.   

Правда ли, что реально нужно просто это причесать и сделать внятную документацию с примером на help.trikset.com ?
Или нужно совсем иное?

iakov, 2021-01-04:
Надо возродить пример, положить все нужные скрипты в инсталлятор тоже, чтобы они были у тех, кто скачал и отставил версию. И сделать документацию. Когда техническую часть сделаем, скинем для документирования.
Технически надо показать (у нас примерно было уже сделано), как можно удалённо по сети вызывать чудо-скрипт для установки/обновления.

iakov, 2021-06-08:
@IKhonakhbeeva , давай добавим всё нужное в инсталлятор. На маке -- в Resources/, на линуксе и винде -- просто в каталог установки. Пусть там будет папка `autoinstall`, в которой все нужные для всех платформ скрипты.
Внимание! Скрипт для винды надо усовершенствовать: если он будет лежать где-то в подкаталоге удаляемого каталога Студии, то не получится удалить каталог со старой студией. То есть в скрипте надо проверять, что он "не в %TEMP%" сейчас, и в этом случае копировать папку `autoinstall` в `%TEMP%`, и вызывать отдельным процессом уже копию скрипта из `%TEMP%`. Но это всё потом, пока просто запакуем в инсталлятор, чтобы было под рукой у всех.

iakov, 2025-02-24:
Tasks.
1. Update and improve scripts (including .cmd).
2. Use our scripts in CI
3. Pack to installer into corresponding folder like `support` or something else for all platforms