trikset/trik-studio #1163: TRIK simulator integration for IDEs doesn't show errors

evsey9, 2020-11-11:
The tool to run 2D models from other IDEs does not show errors. Using the CMD file that I edited myself: https://github.com/evsey9/improved-trik-external-simulation-tool 
Running TRIK Studio 2020.5, Win10 x64, running 2D model from PyCharm Community Edition 2020.2

anastasiia-kornilova, 2021-01-04:
@evsey9 , could you specify in more detail what behavior you expect? Preferably with examples.

evsey9, 2021-01-19:
> 
> 
> @evsey9 , could you specify in more detail what behavior you expect? Preferably with examples.

the behavior I expect is the simulator showing robot's console and error output at the bottom of the screen, like in TRIK Studio simulator, which it doesn't do right now, meaning that if a runtime error happens - you wouldn't even be able to know how and where it happened.

evsey9, 2021-02-20:
issue fixed by adding --console flag to the TRIK simulator tool script

closed 2021-02-20 by evsey9