trikset/trik-studio #1228: TRIK_PYTHONPATH

flvv, 2021-01-06:
MacOS Mojave а так же BigSur 
При попытке запустить Python код в результате ошибка:

`[02:13:24]: ОШИБКА: Для запуска программы на Python доложна быть корректно выставлена переменная окружения TRIK_PYTHONPATH`

Посмотрел все переменные окружения с помощью **printenv** _(TRIK_PYTHONPATH отсутствует)_
Попытался добавить вручную по примеру: https://dl.trikset.com/ts/example.txt но увы все та же ошибка.

flvv, 2021-01-07:
Немного поломав голову все же нашел решение: 

1. Скачиваем архив https://dl.trikset.com/ts/python3.9_homebrew_bigsur.zip 
2. Переходим в директорию где расположена программа «TRIK Studio». В моем случае это стандартный путь при установке: Users/имяпользователя/TRIKStudio/
3. Нажимаем правой кнопкой мыши по TRIK Studio выбираем «Показать содержимое пакета»
4. Переходим Contents/Resources/ и в эту папку распаковываем ранее скачанный архив python3.9
5. В терминале операционной системы вводим nano ~/.bash_profile нажимаем enter
6. Добавляем в конец строку env TRIK_PYTHONPATH=$HOME/TRIKStudio/TRIK\ Studio.app/Contents/Resources/python3.9 open ~/TRIKStudio/TRIK\ Studio.app и сохраняем файл
7. Выполняем перезагрузку

Только это не идеальный способ. Так как если закрыть программу все настройки сбросятся. И чтобы перезапустить программу придется перезагружать сам компьютер.

iakov, 2021-01-07:
Спасибо большое за детальную инструкцию. Да, сейчас есть такая проблема. К сожалению, из-за большого количества запросов поддержка десктоп-версии для Линукс и версии макОС (даже в объёме ТРИК) немного отстают. Поддержку Python "из коробки" мы рассчитываем сделать в ближайшей версии. Подробнее вот тут: #878

closed 2021-01-07 by iakov