trikset/trik-studio #1632: How to link with the quazip and qscintilla system libraries

blackcross86, 2023-02-22:
RosaLinux
version 2022.2

How to link system libraries to avoid re-assembly of quazip and qscintilla.The project is being assembled, but it's getting in the way.

iakov, 2023-03-01:
TL;DR: no chance in 2022.2, but some improvements are in `master` already, I will kindly appreciate your feedback.

Full answer.
By some historical reasons (cross-platformness problems, multiplied by obsolete macOSes and Win32 targets) TS uses it's own versions of these libraries.
`QuaZip` has improved, and now there is `CONFIG+=use_system_quazip` qmake option that can be used. Please, check if this works fine for ROSA. We are ready to improve TS build for Linux distros. Feel free to open a separate ROSA-build related issue. 

`QScintilla` problem is different. Firstly, TS contains a patched version of QScintilla. Probably, these patches are useless after QScintilla is updated. However, until TS updates to the latest QScintilla, it is close to impossible to state this. Secondly, there was an attempt to introduce system QScintilla, and I've refreshed it (follow #1621, maybe it will be merged soon).

blackcross86, 2023-03-01:
```patch
diff --git a/installer/packages/qreal-base/ru.qreal.root/meta/prebuild-common.sh b/installer/packages/qreal-base/ru.qreal.root/meta/prebuild-common.sh
index 0d04a7f..733222b 100755
--- a/installer/packages/qreal-base/ru.qreal.root/meta/prebuild-common.sh
+++ b/installer/packages/qreal-base/ru.qreal.root/meta/prebuild-common.sh
@@ -11,8 +11,6 @@ rsync -a "$INSTALLER_ROOT"/licenses/OFL*.txt                                 "$P
 rsync -a "$INSTALLER_ROOT"/licenses/QsKineticScroller-license*.txt           "$PWD"
 rsync -a "$INSTALLER_ROOT"/licenses/QsLog-license*.txt                       "$PWD"
 rsync -a "$INSTALLER_ROOT"/licenses/QtPropertyBrowser-license*.txt           "$PWD"
-rsync -a "$INSTALLER_ROOT"/licenses/QScintilla-GPL-exception*.txt            "$PWD"
-rsync -a "$INSTALLER_ROOT"/licenses/Scintilla-license*.txt                   "$PWD"
 rsync -a "$INSTALLER_ROOT"/licenses/WindowsModernStyle-license*.txt          "$PWD"
 rsync -a "$INSTALLER_ROOT"/licenses/DejaVu_Fonts_License*.txt                "$PWD"
 rsync -a "$INSTALLER_ROOT"/licenses/acknowledgements.txt                     "$PWD"/../data
diff --git a/installer/packages/qreal-base/ru.qreal.root/meta/prebuild-linux-gnu.sh b/installer/packages/qreal-base/ru.qreal.root/meta/prebuild-linux-gnu.sh
index 1590539..f5e811b 100755
--- a/installer/packages/qreal-base/ru.qreal.root/meta/prebuild-linux-gnu.sh
+++ b/installer/packages/qreal-base/ru.qreal.root/meta/prebuild-linux-gnu.sh
@@ -20,7 +20,6 @@ rsync -a "$BIN_DIR"/libqrgraph.so*                                          "$PW
 rsync -a "$BIN_DIR"/libqrrepo.so*                                           "$PWD"/../data/lib/
 rsync -a "$BIN_DIR"/libqrutils.so*                                          "$PWD"/../data/lib/
 rsync -a "$BIN_DIR"/libqrtext.so*                                           "$PWD"/../data/lib/
-rsync -a "$BIN_DIR"/libqscintilla2.so*                                      "$PWD"/../data/lib/
 rsync -a "$BIN_DIR"/libtrikQsLog.so*                                        "$PWD"/../data/lib/
 rsync -a "$BIN_DIR"/libqrgui-brand-manager.so*                              "$PWD"/../data/lib/
 rsync -a "$BIN_DIR"/libqrgui-controller.so*                                 "$PWD"/../data/lib/
diff --git a/plugins/robots/checker/scripts/build-checker-installer.sh b/plugins/robots/checker/scripts/build-checker-installer.sh
index cfa973d..b6a873d 100755
--- a/plugins/robots/checker/scripts/build-checker-installer.sh
+++ b/plugins/robots/checker/scripts/build-checker-installer.sh
@@ -123,7 +123,6 @@ $qRealDir/libqrkernel.so* \
 $qRealDir/libqrrepo.so* \
 $qRealDir/libqrtext.so* \
 $qRealDir/libqrutils.so* \
-$qRealDir/libqscintilla2.so* \
 $qRealDir/libqrtext.so* .
 
 # Copying TRIKStudio plugins
diff --git a/plugins/robots/generators/ev3/ev3RbfGenerator/ev3RbfGenerator.pro b/plugins/robots/generators/ev3/ev3RbfGenerator/ev3RbfGenerator.pro
index 60ad1df..3023dac 100644
--- a/plugins/robots/generators/ev3/ev3RbfGenerator/ev3RbfGenerator.pro
+++ b/plugins/robots/generators/ev3/ev3RbfGenerator/ev3RbfGenerator.pro
@@ -32,7 +32,7 @@ includes(plugins/robots/generators/ev3/ev3GeneratorBase \
 		qrtext \
 )
 
-links(qrkernel qrutils qrrepo qscintilla2 qrgui-text-editor qrtext \
+links(qrkernel qrutils qrrepo qscintilla2_qt5 qrgui-text-editor qrtext \
 		robots-generator-base robots-ev3-generator-base robots-ev3-kit robots-kit-base robots-utils)
 
 TRANSLATIONS = $$PWD/../../../../../qrtranslations/ru/plugins/robots/ev3RbfGenerator_ru.ts
diff --git a/plugins/robots/generators/nxt/nxtOsekCGenerator/nxtOsekCGenerator.pro b/plugins/robots/generators/nxt/nxtOsekCGenerator/nxtOsekCGenerator.pro
index 798a25a..1cde20e 100644
--- a/plugins/robots/generators/nxt/nxtOsekCGenerator/nxtOsekCGenerator.pro
+++ b/plugins/robots/generators/nxt/nxtOsekCGenerator/nxtOsekCGenerator.pro
@@ -31,7 +31,7 @@ includes(plugins/robots/generators/nxt/nxtGeneratorBase \
 		qrtext \
 )
 
-links(qrkernel qrutils qrrepo qscintilla2 robots-utils robots-nxt-kit \
+links(qrkernel qrutils qrrepo qscintilla2_qt5 robots-utils robots-nxt-kit \
 		robots-generator-base robots-nxt-generator-base)
 
 TRANSLATIONS = \
diff --git a/plugins/robots/generators/nxt/nxtRussianCGenerator/nxtRussianCGenerator.pro b/plugins/robots/generators/nxt/nxtRussianCGenerator/nxtRussianCGenerator.pro
index 9fec79d..8bb7da7 100644
--- a/plugins/robots/generators/nxt/nxtRussianCGenerator/nxtRussianCGenerator.pro
+++ b/plugins/robots/generators/nxt/nxtRussianCGenerator/nxtRussianCGenerator.pro
@@ -29,7 +29,7 @@ includes(plugins/robots/generators/nxt/nxtGeneratorBase \
 		qrtext \
[... 301 lines cut ...]
--- a/qrtest/unitTests/pluginsTests/robotsTests/interpretersTests/support/mockKitPlugin2/mockKitPlugin2.pro
+++ b/qrtest/unitTests/pluginsTests/robotsTests/interpretersTests/support/mockKitPlugin2/mockKitPlugin2.pro
@@ -21,7 +21,7 @@ DESTDIR = $$DESTDIR/plugins/unittests/
 
 QT += widgets
 
-includes(thirdparty/qscintilla/Qt4Qt5 plugins/robots/common/kitBase plugins/robots/utils qrgui qrtext)
+includes(/usr/include/qt5/Qsci plugins/robots/common/kitBase plugins/robots/utils qrgui qrtext)
 
 links(robots-kit-base)
 
diff --git a/thirdparty/thirdparty.pro b/thirdparty/thirdparty.pro
index 81857c5..99ef0e8 100644
--- a/thirdparty/thirdparty.pro
+++ b/thirdparty/thirdparty.pro
@@ -18,7 +18,6 @@ HEADERS += \
 	runExtensions.h \
 
 SUBDIRS = \
-	qscintilla/Qt4Qt5/qscintilla.pro \
 	qslog/QsLogSharedLibrary.pro \
 	gamepad/gamepad.pro \
 	quazip/quazip.pro \

```

iakov, 2023-03-01:
Usually, few steps are enough, if corresponding packages were installed in advance. Like `mkdir build && cd build && qmake ../studio/studio.pro CONFIG+=ccache CONFIG+=release CONFIG+=force_debug_info ...` .
If something does not work, let's check exact problems: what you expect, and what is happening.

blackcross86, 2023-03-01:
> Usually, few steps are enough, if corresponding packages were installed in advance. Like `mkdir build && cd build && qmake ../studio/studio.pro CONFIG+=ccache CONFIG+=release [...]

Everything came together successfully.Starting the installation, where should it all be installed? Example: gamepad is not displayed in the startup menu.

iakov, 2023-03-01:
External utilities are to be used from `Tools -> External tools` menu, and are configured via `externalTools.xml`. BTW, do not expect `make install` to work as usually, please, it is totally broken.

blackcross86, 2023-03-01:
> External utilities are to be used from `Tools -> External tools` menu, and are configured via `externalTools.xml`. BTW, do not expect `make install` to work as usually, please, it is totally broken.

make install, as I already understood, all folders and files are scattered. But it is not clear where to install, there are installation directories, what is the location of files in your program for full use?

iakov, 2023-03-01:
Well... Not a simple question, actually. Please, take a look into the [QtIFw packaged installer](https://dl.trikset.com/ts/fresh/installer/trik-studio-installer-linux-2022.2.run)/installation. Or check AltLinux packaging "hacks". 

TS has patched RPATH, and expects 'lib' folder (AFAIR) and some not obvious folders structure. Since this has to be fixed, please, feel free to mention your distro problems. 

Also, there is a `trik-studio.ini` by QSettings, that can be used to configure paths (for plugins, resources, etc.), probably, `/etc/xdg/.../trik-studio.ini` can be used to override. Take a look at [platform-specific section](https://doc.qt.io/qt-6/qsettings.html#platform-specific-notes) about `/etc/xdg`/`$XDG_CONFIG_DIRS`
This screenshot shows search paths on Ubuntu:
[image]
Thus, this `.ini` can be used to override paths, and the `.ini` itself can be found after the first run in users `.config` folder.

blackcross86, 2023-03-01:
Altlinux has a crooked project, just like me.But thank you all the same for the explanations.

iakov, 2023-03-02:
Let's stay in touch for further TS improvements. Thanks to AltLinux, some improvements generally useful for distro packaging have been upstream-ed already.

iakov, 2024-12-07:
@blackcross86 , 2024.1 is ready. Lot of refactorings and build system / CI improvements in this release, thus you are welcome to report any issues. If you are the maintainer of the corresponding package in any distro, or a amatuer TS user, let's try to impove build system toward your needs together.
Also, I'm interested and ready to take a sight at your fixes, maybe we can help to simplify furter maintanance. I've created [this issue ](https://github.com/trikset/trik-studio/issues/1843) to maintain a list of contributions for futher integration.
FYI: 2024.1 has `CONFIG+=use_system_qscintilla2` and `CONFIG+=use_system_quazip` options for `qmake` . Maybe this helps.

closed 2024-12-07 by iakov