trikset/trik-studio #1654: python 3.11 throws an error.

blackcross86, 2023-04-19:
```python3

/usr/include/python3.11/object.h:107:28: note: in expansion of macro '_Py_CAST'
  107 | #define _PyObject_CAST(op) _Py_CAST(PyObject*, (op))
      |                            ^~~~~~~~
/usr/include/python3.11/object.h:136:31: note: in expansion of macro '_PyObject_CAST'
  136 | #  define Py_TYPE(ob) Py_TYPE(_PyObject_CAST(ob))
      |                               ^~~~~~~~~~~~~~
/usr/include/python3.11/tupleobject.h:27:38: note: in expansion of macro 'Py_TYPE'
   27 |                  PyType_FastSubclass(Py_TYPE(op), Py_TPFLAGS_TUPLE_SUBCLASS)
      |                                      ^~~~~~~
PythonQt.cpp:2361:26: note: in expansion of macro 'PyTuple_Check'
 2361 |                 Q_ASSERT(PyTuple_Check(code->co_varnames));
      |                          ^~~~~~~~~~~~~
PythonQt.cpp:2363:58: error: 'PyCodeObject' {aka 'struct PyCodeObject'} has no member named 'co_varnames'; did you mean 'co_names'?
 2363 |                   PyObject* name = PyTuple_GetItem(code->co_varnames, i);
      |                                                          ^~~~~~~~~~~
      |                                                          co_names
PythonQt.cpp:2368:55: error: 'PyCodeObject' {aka 'struct PyCodeObject'} has no member named 'co_varnames'; did you mean 'co_names'?
 2368 |                   PyObject* s = PyTuple_GetItem(code->co_varnames, nargs);
      |                                                       ^~~~~~~~~~~
      |                                                       co_names
PythonQt.cpp:2374:55: error: 'PyCodeObject' {aka 'struct PyCodeObject'} has no member named 'co_varnames'; did you mean 'co_names'?
 2374 |                   PyObject* s = PyTuple_GetItem(code->co_varnames, nargs);
      |                                                       ^~~~~~~~~~~
      |                                                       co_names
make[6]: *** [Makefile:645: .build/x86-release/obj/PythonQt.o] Error 1
make[6]: Leaving directory '/root/trik-studio/BUILD/trik-studio-2022.2/plugins/robots/thirdparty/trikRuntime/trikRuntime/PythonQt/PythonQt/src'
make[5]: *** [Makefile:48: sub-PythonQt-src-make_first] Error 2
make[4]: *** [Makefile:52: sub--root-trik-studio-BUILD-trik-studio-2022-
```
Rosa Linux

Python 3.11 causes the build to fail.

iakov, 2023-04-21:
Thanks for reporting. This is the same as https://github.com/trikset/trik-pythonqt/issues/14

blackcross86, 2023-04-21:
Yes, thanks.The build was successful

iakov, 2024-10-23:
Fixed.
Python versions up to 3.12 are supported, and the tagged release is expected to be 2024.1.
However, `qmake` options have changed, see GH workflows for up to date examples.

closed 2024-10-23 by iakov